cujomalainey/ant-arduino #53: Guidance project

ketyas, 2020-07-21:
Hello,

What exacly i need to built a heart ratę monitor? Its arduino and nrf52 chip enought?

cujomalainey, 2020-07-21:
Hello @ketyas 

Yes, you can connect to an ANT+ HR strap if you buy just the [Adafruit nRF52 Express](https://www.adafruit.com/product/4062). But you must change the softdevice on the board which requires changing the bootloader. This requires fetching the new softdevice from [dynastream](https://www.thisisant.com/developer/components/nrf52832/). Changing the bootloader (there are guides) is a more complex/riskier process because if not done correctly can brick the board and require a [programmer](https://www.adafruit.com/product/3571) to restore the board.

The alternative would be to use an external [D52 module](https://www.digikey.ca/en/products/detail/garmin-canada-inc/D52QD2M6IA-A-TRAY/6149453) via Serial. This is lower risk since it requires no changes to bootloader but will require additional wiring between the arduino and the D52 module. If you are willing to do a custom PCB order you can buy a [ANT-Featherwing](https://github.com/reeganm/ANT-FeatherWing) from a PCB manufacturer and use a compatible feather board (check pins first, not all boards are compatible [e.g. particle argon])

ketyas, 2020-07-22:
Thanks for explaining so thoroughly.  Now I know how to do it.

closed 2020-07-22 by ketyas

Abasz, 2022-08-16:
Hi, I have this [NRF52840-DONGLE](https://www.tme.eu/details/nrf52840-dongle/adatatviteli-fejlesztoeszkozok/nordic-semiconductor/?brutto=1&currency=HUF&gclid=Cj0KCQjwgO2XBhCaARIsANrW2X1phALDv7K6IzQcjzyIUiumTATOJAzn-geY6qvrEFtXc4gYqQtnt6waApxKEALw_wcB) that I would like to use with my ESP32 to add Ant+ capabilities, but I am struggling to make it work. 

As far as I understand the DONGLE comes with the BLE only SoC so I need to update the softdevice to use S340 (BLE+ANT). But does your library work with S212 (i.e. ANT only)? I don't need BLE as I have that already working trough the ESP.

My second question is, whether I need to upload any code to the DONGLE (apart from the bootloader and the correct softdevice of course) to enable serial (UART or SPI) communication between ESP and the DONGLE with your library.

thanks in advance

cujomalainey, 2022-08-17:
Hi @Abasz 

The library works with any softdevice that supports ANT. There is a caveat though. When using the radio via UART or SPI you need to load it with the [network processor firmware](https://www.thisisant.com/developer/resources/downloads/#software_tab) which does have restrictions on the softdevice to use. The reason I recommend the D52's is because they come pre-loaded with this firmware so its just easier.

I think that also answers your second question. Let me know if you need any more help.

Abasz, 2022-08-19:
Thanks for the prompt reply. I think I understand now the issue and the main reason for recommending D52s. My problem with those is that I am not able to find one in my region and importing with shipping would cost $50+...

But I think I understand what needs to be done: 
1. Have to check which softdevices are supported by the ANT Network Protocol Firmware (as far as I can tell from the thisisant page that these are: s210, s212, s310, s332 - although only s210 and s310 comes precompiled)
2. Need to check if the NRF52840 chip supports any of these softdevices (unfortunately I think no...)
3. Need to check which NRF52 chip does support this (I think nRF52832 supports s212)
4. Need to compile/load the nRF52832 chip with the s212 softdevice along with the NRF52 Network Processor Source code

Does the above sound reasonable? I am not sure this would be feasible but at least from a theoretically point of view :)

cujomalainey, 2022-08-22:
Yep that sounds exactly what you need to do. If you have a programmer handy, it shouldnt be too hard to achieve.