cujomalainey/ant-arduino #56: NRF82532 / NRF52840

tcpipchip, 2022-03-26:
Hi
I have here those models and ANT USB dongle to PC

Will work with them ?

cujomalainey, 2022-03-27:
Hi,

Yep you can communicate over ANT to your PC, you just have to make sure the nRF devices are loaded with an ANT enabled soft device. https://thisisant.com keeps a compatibility list and you can download the new softdevices from there. Most ship with only BLE soft devices.

tcpipchip, 2022-03-27:
Nice, I will test, make more blogs and promote you GitHub ANT :)

https://loja.smartcore.com.br/pagina/blog-faq-familia-nina-u-blox-wifi-ble.html

cujomalainey, 2022-03-27:
Awesome :) let me know if you need any more info, @orrmany wrote an awesome guide on switching to the S340 on the adafruit feathers

tcpipchip, 2022-03-27:
Please,send me the guide!

cujomalainey, 2022-03-27:
Here is the blog post https://blogarak.wordpress.com/2020/03/15/s340-softdevice-adafruit-nrf52840-express-feather/

For sending sensor values are you looking to use ANT+ or just raw data (if ANT+ see my antplus-arduino lib)

tcpipchip, 2022-03-27:
About the blog
Why recompile bootloader if you can download the ant softdevice hex later ?

At moment, send raw data using ant+, or am i am confused about concept?

cujomalainey, 2022-03-27:
On adafruit boards the soft device and bootloader are part of the same region when updated so you have to do both at the same time. When you update the application the softdevice is not updated. Also the bootloader needs to be updated as it needs to power up the soft device initially so needs the new API calls if I recall correctly.

And ANT+ is merely a protocol for ANT, you in theory can define what ever byte pattern you want over the radio system

tcpipchip, 2022-03-28:
Nice
About network,when I ordered 4 years ago my usb dongle, must i have rights to network key ? If not, how can I get it ?

cujomalainey, 2022-03-28:
You can get the ANT+ network key from http://www.thisisant.com/developer/ant-plus/ant-plus-basics/network-keys/

You just have to agree to their "adopter terms" first. I recommend you take a quick read as that is why I do not have they key in any of my public repos.

tcpipchip, 2022-03-29:
hi
wich is the difference between 
LICENCE_KEY and 
NETWORK KEY ?

On adopt, i got network key
But, and what is network KEY ?

On Arduino examples, will i need both ?

Thanks for all help!

cujomalainey, 2022-03-30:
The license key is used for the softdevice itself. The headers should include an evaluation license key that you can just uncomment. This simply allows you to use the code.

The network key is an 8 byte number that identifies a wireless network in ANT. The one provided at the link above has the keys for ANT+, ANT-FS and the public key.

You will need both, the license key to enable the software, and the network key to join the correct networks

tcpipchip, 2022-03-30:
In my case, using segger studio, and uncommented the ANT_LICENSE_KEY, i enter in a ANT network, where there is not ANT+ products running...right ? In theory, not for commercial use

To make commercial use (create a product), need to enter in the ANT+ network, right, using the Network Key ? Or enter in ANT+ network, created by end-devices, as Garmir GPS...right ?

You Arduino LIB uses ANT , not ANT+, right ?

For example, i followed the sequence of this NORDIC INSTRUCTIONS

https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v15.3.0%2Fant_examples_broadcast.html&cp=5_1_4_0_3
Just uncomment that line...burn...and run, works

In you Arduino, i have to modify to the following 

Channel type	Master (0x10)	
Network key	Public	
RF channel	66 (2466 MHz
Device number	0x02	
Device type	0x02	
Transmission type	0x01
Channel period	8192 (4 Hz)	

How can in set this on Arduino ?

Thank you about all answer, this will help all reading this DOC

tcpipchip, 2022-03-31:
Hi, 
No commercial, just to show to u-blox readers that is possible to use ANT on NINA B112...
https://nina-b3-openthread.blogspot.com/2022/03/u-blox-nina-b112-transmitindo-dados-via.html
Next step is use your ANTPLUS-ARDUINO, but can i test it with ANTWare Dongle ?

cujomalainey, 2022-04-01:
If you mean can you mock a fake device with a USB dongle and Simulant+, the answer is yes. That is how I do most of my testing.

tcpipchip, 2022-04-01:
hi ant-plus project still compiling on VISUAL STUDIO CODE ?
https://github.com/cujomalainey/antplus-arduino

cujomalainey, 2022-04-01:
yep, should compile in platformio or the arduino editor. I use the platformio CLI personally. Are you seeing issues?

tcpipchip, 2022-04-02:
Yes, issues! Tested on 3 machines! Using pkatformio

cujomalainey, 2022-04-02:
> Yes, issues! Tested on 3 machines! Using platformio 
> May I send you late ?

I'm guessing you mean logs? And sure you can post them here or email them to me.

tcpipchip, 2022-04-02:
step by step
[image]
[image]
[image]
[image]
[image]

tcpipchip, 2022-04-02:
i had to modify to have some progress
[env:nrf52_express]
board = adafruit_feather_nrf52832
framework = arduino
platform = nordicnrf52
platform_packages = framework-arduinoadafruitnrf52
lib_deps = https://github.com/cujomalainey/ant-arduino.git#develop

[image]

tcpipchip, 2022-04-02:
back to 832 and remove all examples, except one!
[image]

tcpipchip, 2022-04-02:
weird problem
[image]

undefined reference to arduino loop many times ?

cujomalainey, 2022-04-02:
so your first screenshot is capturing a bug in platformio itself, you need to search the forums for that.

Your following screenshots are because you are using platformio and vscode like the arduino editor. The arduino editor is a very limited tool and it only compiles what you see. Almost every other editor in the world does not do this. 

Two things you need to do to solve your problem.
1. make a new project. I can see you currently have my repository open. Unless you plan to contribute back to my project this is not what you want to do.
2. add antplus arduino as a dependency via `lib_deps = ANT+` you don't need to worry about ant-arduino as it is a dependency for the antplus package.
3. copy an example you wish to run into your src folder. Platformio under normal build conditions will only build what is in src/ and lib/

tcpipchip, 2022-04-03:
Hi
On NRF52840,  i had to include
#include "Adafruit_TinyUSB.h"
Getting progress!

cujomalainey, 2022-04-03:
Interesting, also you need to use the arduinonativeantwithcallbacks if you are using the in device radio

tcpipchip, 2022-04-03:
Do you mean that this example use a external Uart art+ radio ? And not the IN radio ?

cujomalainey, 2022-04-03:
Yes, see the native ant example in ant Arduino to see how to set it up

tcpipchip, 2022-04-03:
well, that example still didnt help me!

Still tryng to understand to make the blog

tcpipchip, 2022-04-03:
same problem on 
Arduino (IDE)

cujomalainey, 2022-04-03:
That means then that your library is not building with the -DS340 flag (assuming you used the S340 softdevice). You need to add that so you have access to the direct soft device APIs. They are behind a build flag otherwise the system would never compile for remote targets using the serial driver.

tcpipchip, 2022-04-04:
then the
ArduinoNativeAnt Class is part of SoftDevice 340 ?

cujomalainey, 2022-04-04:
Not quite, its a wrapper for the S340 API. The Serial driver is designed for the network processor where the nRF52 is external chip to the one you are programming. ant-arduino is meant to provide a uniform interface regardless how you interface with the radio.

tcpipchip, 2022-04-05:
@cujomalainey 

Can you zip us a ant+ platformio project ready to download ?

cujomalainey, 2022-04-05:
@orrmany thanks for the headsup, I will try and rebase my fork and see I can get it working with the latest BSP. Bootloader should be compatible regardless version.

@tcpipchip I cannot, distributing the license key and the ANT headers is against the agreement. We can setup the project but you have to add the files from dynastream.

cujomalainey, 2022-04-06:
I fixed up the framework, https://github.com/cujomalainey/Adafruit_nRF52_Arduino now its based off the latest upstream tag

I just have to push a platform package as well with the board definitions

I am also going to going to generate a ant-arduino reference in platformio as you are not the first person this week confused by this convoluted workflow :)

cujomalainey, 2022-04-08:
Got a bit further, set up the base [platform](https://github.com/cujomalainey/platform-nordicnrf52) with a feather board reference. Although I have hit some problems with the adafruit nrf crypto library and how to properly include it, not sure why its breaking. I am going offline for the weekend, will pick up next week.

cujomalainey, 2022-04-16:
Sorry for the radio silence, still working on the crypto library bug.

cujomalainey, 2022-04-16:
Here is a buildable config for platformio.
```
platform = https://github.com/cujomalainey/platform-nordicnrf52#develop
board = adafruit_feather_nrf52840_s340
framework = arduino
lib_deps = ANT
build_flags = -DS340
```

Still need to verify against my board.

The `build_flags` is temporary till https://github.com/platformio/platform-nordicnrf52/issues/147 is resolved

cujomalainey, 2022-04-27:
Please reopen if you have anymore questions or comments

closed 2022-04-27 by cujomalainey

orrmany, 2022-10-11:
I think tiny-usb is also changed a lot and that is very relevant to the
Adafruit bootloader. I plan to restart my work on my ANT+ project, too, so
I will make an attempt of catching up my fork with upstream, too. I had to
change computer meanwhile, so currently my PlatformIO instance is broken,
too, i.e.: don't expect overnight update from me :P

Curtis Malainey ***@***.***> ezt írta (időpont: 2022. ápr.
5., K, 18:58):

tcpipchip, 2022-10-11:
this is very very cool news

cujomalainey, 2022-10-11:
Jeez we are at 9.3 already? Thanks @orrmany for taking the lead. My focus has been on my ANT rust crate, hoping to publish by EOY. Let me know when you have it working and I can make sure my stuff still works.

cujomalainey, 2022-12-02:
FYI i rebased my branches, but I haven't be able to test as my board refuses to be programmed (might have to find my jlink).

tcpipchip, 2022-12-03:
NICE

orrmany, 2022-12-04:
Got to a state, where I became able to rebuild the current stock master version of the Adafruit Bootloader in PlatformIO, that is, I've completed Step #1 of 9 steps of https://blogarak.wordpress.com/2020/03/29/platformio-ide-integration-for-the-nrf52840-feather-express-with-s340/#adding_s340_support_to_plaform_io 
Then I started to "port" my ANT softdevice changes to the new version. Currently I am stuck at #6, modifying the linker --- then I had to realize that the 0.4.0 release version is a major change: Ha Tach has removed the softdevice dependency from the bootloader, see https://github.com/adafruit/Adafruit_nRF52_Bootloader/blob/master/changelog.md#040---20210119

Now I am way out of my competence with this bootloader stuff right now, so it will take more time than I anticipated. :(

cujomalainey, 2022-12-04:
Ah i spent my time fighting with the bsp, I didn't even attempt the bootloader. Indeed this changes a lot.

Thankfully it reduces risks of mistakes as we can just say "update with a prebuilt" (assuming the bootloader doesn't need an alternative linker script). But it indeed throws out almost our work for loading the softdevice.