cujomalainey/ant-arduino #58: Quick start hardware recommendations

kostrse, 2022-12-03:
For someone who didn't deal with nRF radios before, it's overwhelming at the start to figure out what is the best development setup to start from.

Could you please share some specific recommendations (a shopping list) in the README what boards and tools to have?

There're have been several SoCs and development boards from Nordic, some of them are already deprecated, the ANT forum is not active as well.

* Is Arduino Due a good candidate for development/prototyping (3.3V, powerful CPU, 4 UARTs, a lot of PINs)
* Raspberry Pi Pico 2040 (3.3V, dual-core CPU, 2 UARTs, also should support PlatformIO)?
* What nRF radio is the best to interface with the microcontroller via UART?

Thanks!

cujomalainey, 2022-12-04:
I will point you to this [comment](https://github.com/cujomalainey/ant-arduino/issues/53#issuecomment-662072762) as a kickstarter.

If you are doing a raspberry Pi you can skip this library and go straight to the libraries provided by dynastream and use a USB stick (but apparently they are super marked up right now [$50 !?!?!?]).

D52's (see link)  as they come preloaded with the correct firmware for serial access. The pin mapping is not the best documented so I recommend checking out my [wiki](https://github.com/cujomalainey/ant-arduino/wiki/Developer's-Guide) for a reference for wiring

cujomalainey, 2022-12-04:
Forgot to mention, any board with 3.3V logic and an external UART is usually a good candidate for development. I personally use the ESP32 usually.

cujomalainey, 2022-12-12:
If you have any more questions please reopen :)

closed 2022-12-12 by cujomalainey